borglab/GTDynamics #266: sdfformat on Ubuntu

dellaert, 2021-09-07:
I can't get sdformat to install on Linux with instructions on website.  I did:

```bash
apt-get -y install libtinyxml2-dev libxml2-utils pkg-config ruby-dev ruby wget
apt-get -y install libignition-math4-dev
cd /usr/src
wget http://osrf-distributions.s3.amazonaws.com/sdformat/releases/sdformat-10.5.0.tar.bz2
tar -xvjf sdformat-10.5.0.tar.bz2
rm -f sdformat-10.5.0.tar.bz2
mkdir build
cd build
cmake ..
```

But get

```
====== Finding 3rd Party Packages ======
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Building for arch: x86_64
-- Found TinyXML2: /usr/lib/x86_64-linux-gnu/libtinyxml2.so  
-- Checking for module 'urdfdom>=1.0'
--   No package 'urdfdom' found
CMake Warning at cmake/SearchForStuff.cmake:34 (find_package):
  By not providing "Findurdfdom.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "urdfdom", but
  CMake did not find one.

  Could not find a package configuration file provided by "urdfdom" with any
  of the following names:

    urdfdomConfig.cmake
    urdfdom-config.cmake

  Add the installation prefix of "urdfdom" to CMAKE_PREFIX_PATH or set
  "urdfdom_DIR" to a directory containing one of the above files.  If
  "urdfdom" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:116 (include)

-- Couldn't find urdfdom >= 1.0, using internal copy
CMake Warning at cmake/SearchForStuff.cmake:55 (find_package):
  By not providing "Findignition-tools.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ignition-tools", but CMake did not find one.

  Could not find a package configuration file provided by "ignition-tools"
  with any of the following names:

    ignition-toolsConfig.cmake
    ignition-tools-config.cmake

  Add the installation prefix of "ignition-tools" to CMAKE_PREFIX_PATH or set
  "ignition-tools_DIR" to a directory containing one of the above files.  If
  "ignition-tools" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:116 (include)

-- 	ignition-tools not found, for command line utilities and for generating aggregated SDFormat descriptions for sdformat.org, please install ignition-tools.
-- Could NOT find PY_psutil (missing: PY_PSUTIL) 
-- 	Python psutil package not found. Memory leak tests will be skipped
-- Found ruby executable: /usr/bin/ruby
CMake Error at cmake/SearchForStuff.cmake:107 (find_package):
  By not providing "Findignition-cmake2.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ignition-cmake2", but CMake did not find one.

  Could not find a package configuration file provided by "ignition-cmake2"
  (requested version 2.3) with any of the following names:

    ignition-cmake2Config.cmake
    ignition-cmake2-config.cmake

  Add the installation prefix of "ignition-cmake2" to CMAKE_PREFIX_PATH or
  set "ignition-cmake2_DIR" to a directory containing one of the above files.
  If "ignition-cmake2" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:116 (include)

-- Configuring incomplete, errors occurred!
```

I can update the README, but can someone comnment on how they got past this? I hope it does not involve also installing urfdom from source etc...

varunagrawal, 2021-09-07:
Use the homebrew instructions in the README. Homebrew supports both Mac and Linux so it should work perfectly.

dellaert, 2021-09-07:
So, why do we have the Ubuntu instructions if they do not work?

varunagrawal, 2021-09-07:
I believe I used those instructions to initially install SDFormat10 to Ubuntu.

I looked up `urdfdom` and it looks like doing `sudo apt install urdfdom-dev` should be sufficient to get that dependency. I would still prefer to kill the source install instructions though.

dellaert, 2021-09-07:
Installing homebrew in docker is a drag. Can someone help me source install on Linux? what did you do to make it work?

dellaert, 2021-09-07:
*somebody* amongs you must have done this, right? @gchenfc @yetongumich @nrakoski3 ?

gchenfc, 2021-09-07:
I used linuxbrew - tried once install from source and it was a nightmare (lots of dependencies) and I gave up.

varunagrawal, 2021-09-07:
@dellaert what version of Ubuntu are you using?

varunagrawal, 2021-09-07:
@dellaert I made a PR with the updated instructions. I used a docker 18.04 image to test this.

PR #268 (2021-09-08): Updated instructions for source install of SDFormat10
The title. Should resolve #266.
closed 2021-09-08 by dellaert