borglab/GTDynamics #271: Remove JointCoordinateType

varunagrawal, 2021-09-16:
`JointTyped` defines two aliases, `JointCoordinateType` and `JointCoordinate` which is the same as `JointCoordinate`. The majority of the code uses `JointCoordinate` while only `JointLimitFactors.h` uses `JointCoordinateType`.

This kind of aliasing is confusing, and unless someone can convince me with a _compelling_ argument otherwise, we should get rid of `JointCoordinateType`.

varunagrawal, 2021-09-16:
@gchenfc @yetongumich can you please let me know if I am missing something and why JointCoordinateType should not be deleted? We should also take some time to add in the ball and sphere joints soon since that may illuminate the usefulness of `JointTyped` going forward.

gchenfc, 2021-11-06:
Resolved by #291 that removed all these things.

closed 2021-11-06 by gchenfc