borglab/GTDynamics #276: Slow CI as compiles GTSAM

dellaert, 2021-09-18:
Can we instead pull a GTSAM docker image?

varunagrawal, 2021-09-18:
I've actually been actively working with @berndpfrommer to get the nightly packages in a state where we can just pull the pre-built binaries. This should shave off a tremendous chunk of time.

We currently have 2 issues:

1. CppUnitLite is not included in the binaries, so the compiler can't find CppUnitLite to build the tests. I've started the process to have a separate build of CppUnitLite. https://github.com/borglab/gtsam-packaging/issues/15
2. There are not docker images for MacOS yet. There is a workaround, but that process will make the build longer. One suggestion I've had is to have Homebrew support for GTSAM so that Homebrew can just download the bottle (Homebrew term for binary). https://github.com/borglab/gtsam-packaging/issues/13

varunagrawal, 2021-12-31:
I figured this out!!! Atleast for the Linux CI. If we can get homebrew binary support for macos, then I can speed that up as well.

varunagrawal, 2022-04-29:
Here's a solution we should have: https://github.com/features/packages

Basically, after a successful CI run, we have the CI trigger a packaging action which will upload a MacOS binary to the package registry. Once we have this set up, we can update the MacOS CI to pull this binary rather than build from scratch.

closed 2023-03-11 by varunagrawal