borglab/GTDynamics #281: All your robots belong to us!

dellaert, 2021-09-24:
Can we get the legged sdfs from the subT challenge to load?
https://github.com/osrf/subt/wiki/Robots

varunagrawal, 2021-09-25:
You mean "All your robots ARE belong to us!"? 😉 

It should be very doable.

varunagrawal, 2021-09-27:
Here's the `cerberus_anymal_b_sensor_config_1` from the above repo.

[image]

dellaert, 2021-09-27:
Cool. PR for this? BTW #253 does not seem to work with our Docker images, says @nrakoski3 . Can you try it inside Docker? Did I not install SDF correctly?

varunagrawal, 2021-09-27:
What's the error message? That it doesn't work doesn't help me at all :)

varunagrawal, 2021-09-27:
I know what the problem is: @nrakoski3 waited too long to try it.

So the PR to get rid of `wTcom` landed but `roboplot` doesn't have that updated yet. I believe this happened somewhere in this past week since I was able to plot the NAO when I created that PR. I'll push an update to roboplot and this should work.

varunagrawal, 2021-09-27:
Okay, the plotting now works even in the docker image.

PR #289 (2021-11-03): Add models from the DARPA SubT challenge
Fixes #281.
closed 2021-11-03 by varunagrawal