borglab/GTDynamics #282: Change `internal::*Key` to `*Key`

varunagrawal, 2021-09-25:
The current pattern we use to define various keys is for example `gtd.internal.PoseKey`. After using this pattern extensively the past many months, it seems unnecessarily verbose, since to add the keys to factor graphs, we have to type the whole thing in, which defeats the purpose of marking it "internal".
The same thing applies when writing tests.

gchenfc, 2021-11-06:
I know this is really old but was just looking back at old issues and saw this

I think this is actually getting at a higher level design philosophy that's related to #290, which is that the `gtdynamics::internal::XXXKey` design philosophy is to force users to stay within the framework we've prescribed to prevent them shooting their own foot.  My goal with #290 was to continue that philosophy to the factor constructors, because as it stands, several factors require you to use the `internal` namespace to construct them which shouldn't be that way.

The alternative philosophy is to allow everyone to use whatever keys they want, even if it doesn't make sense.  Although this allows us to hack new applications faster, personally I think it's ultimately more cumbersome and prone to bugs that are difficult to debug (correctness bugs).

Just my $0.02

varunagrawal, 2021-11-06:
The issue here is that if I use a GTSAM key, that does not have an index referring to the link or the joint, and so there is a type mismatch.

A simple example for you to try: run a forward kinematics optimization using a GTSAM key and you'll see where these high level functions fail.

It's nice to have these conveniences but not at the cost of flexibility. A user should be able to use whatever key makes semantic sense for their use case.

Regarding the internal, we are using internal::key when adding factors to the factor graph so that basically defeats the purpose.

dellaert, 2021-11-06:
I like both points of view, but I find myself agreeing with @varunagrawal more, so, let's remove the verbose namespace

PR #328 (2022-01-01): Remove `internal` namespace
Fixes #282
closed 2022-01-01 by varunagrawal