borglab/GTDynamics #320: Need for shared pointers are arguments?

varunagrawal, 2021-12-08:
Great stuff on the new updates.

However I had a question about using shared pointers as function and constructor args. Is there a reason we're doing `Optimizer(const boost::shared_ptr<const OptimizationParameters>& parameters)` instead of `Optimizer(const OptimizationParameters& parameters)`?

Seems like unnecessary verbosity to me especially when `OptimizationParameters` is a simple struct (aka an encapsulation of various parameters).

dellaert, 2021-12-08:
ShareD Pointers are great for noise models, which could be used in thousands of factors or even millions. If we only create a handful or one optimizer ever, I agree it is not needed here.

varunagrawal, 2021-12-08:
That's what I figured as well. Highly unlikely we'll need hundreds of optimizer parameters ever.

PR #323 (2021-12-21): Convert Optimizer Arguments from shared pointers to objects
This fixes #320

This PR proposes to swap `Optimizer(const boost::shared_ptr<const OptimizerParams>& params)` with the simpler `Optimizer(const OptimizerParams& params)`. Currently we make use of shared pointers for optimizer parameters which seems unnecessarily verbose since we don't expect to have more than a handful of optimizers running. By using const references, we avoid the need to call `make_shared` at various places and improve the overall readability of the code.

Other benefits include being able to get quick auto-complete  [... cut]
closed 2021-12-21 by yetongumich