borglab/GTDynamics #336: DynamicsGraph.h missing #include <boost/format.hpp>

arutkowski, 2022-02-19:
Trying to build GTDynamics, I got the following error:

```
In file included from /home/adam/GTDynamics/gtdynamics/dynamics/DynamicsGraph.cpp:34:
/home/adam/GTDynamics/gtdynamics/utils/JsonSaver.h: In static member function ‘static std::string gtdynamics::JsonSaver::GetNoiseModel(const shared_ptr&)’:
/home/adam/GTDynamics/gtdynamics/utils/JsonSaver.h:262:22: error: ‘format’ is not a member of ‘boost’; did you mean ‘boost::timer::format’?
  262 |         ss << boost::format("isotropic dim=%1% sigma=%2%") %
      |    
```

Adding `#include <boost/format.hpp>` to `DynamicsGraph.h` seems to fix the issue.

dellaert, 2022-02-19:
Ah, yes - I have removed that include from Matrix.h in gtsam. Sorry!
Feel free to make a PR but that file should be in a cpp file where used if possible to avoid header bloat.

dellaert, 2022-02-19:
I went ahead and created #337 but now get in trouble building python wrapper. Confirm this works for you?

varunagrawal, 2022-02-19:
Can confirm #337 works. @dellaert can you try rebuilding the project from scratch?

dellaert, 2022-02-19:
Removing build and starting from scratch worked. Probably because my build folder was rather old.
I'll close this issue then.

closed 2022-02-19 by dellaert